xArm-Developer/xarm_ros #236: Error while trying to use the xArm6 Moveit package with ROS_NAMESPACE

nitish3693, 2024-07-24:
Hi Team,

When trying to launch the `realMove_exec.launch` from the `xarm6_moveit_config` under a ROS namespace, the process is giving errors and not connecting to the `follow_joint_trajectory` server.

Logs,

```
Ubuntu20:~/ros_wk$ ROS_NAMESPACE=test roslaunch xarm6_moveit_config realMove_exec.launch robot_ip:=192.168.1.212 
... logging to /home/nitish/.ros/log/226b56c6-49d9-11ef-be2f-c9a005e8bc99/roslaunch-Ubuntu20-9361.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Using load_yaml() directly is deprecated. Use xacro.load_yaml() instead.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Ubuntu20:45153/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /source_list: ['xarm/joint_stat...
 * /test/joint_state_publisher/rate: 10
 * /test/joint_state_publisher/use_gui: False
 * /test/move_group/allow_trajectory_execution: True
 * /test/move_group/capabilities: move_group/MoveGr...
 * /test/move_group/controller_list: [{'name': '', 'ac...
 * /test/move_group/jiggle_fraction: 0.05
 * /test/move_group/max_range: 5.0
 * /test/move_group/max_safe_path_cost: 1
 * /test/move_group/moveit_controller_manager: moveit_simple_con...
 * /test/move_group/moveit_manage_controllers: True
 * /test/move_group/octomap_resolution: 0.025
 * /test/move_group/planner_configs/BFMT/balanced: 0
 * /test/move_group/planner_configs/BFMT/cache_cc: 1
 * /test/move_group/planner_configs/BFMT/extended_fmt: 1
 * /test/move_group/planner_configs/BFMT/heuristics: 1
 * /test/move_group/planner_configs/BFMT/nearest_k: 1
 * /test/move_group/planner_configs/BFMT/num_samples: 1000
 * /test/move_group/planner_configs/BFMT/optimality: 1
 * /test/move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /test/move_group/planner_configs/BFMT/type: geometric::BFMT
 * /test/move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /test/move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /test/move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /test/move_group/planner_configs/BKPIECE/range: 0.0
 * /test/move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /test/move_group/planner_configs/BiEST/range: 0.0
 * /test/move_group/planner_configs/BiEST/type: geometric::BiEST
 * /test/move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /test/move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /test/move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /test/move_group/planner_configs/BiTRRT/init_temperature: 100
 * /test/move_group/planner_configs/BiTRRT/range: 0.0
 * /test/move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /test/move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /test/move_group/planner_configs/EST/goal_bias: 0.05
 * /test/move_group/planner_configs/EST/range: 0.0
 * /test/move_group/planner_configs/EST/type: geometric::EST
 * /test/move_group/planner_configs/FMT/cache_cc: 1
 * /test/move_group/planner_configs/FMT/extended_fmt: 1
 * /test/move_group/planner_configs/FMT/heuristics: 0
 * /test/move_group/planner_configs/FMT/nearest_k: 1
 * /test/move_group/planner_configs/FMT/num_samples: 1000
 * /test/move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /test/move_group/planner_configs/FMT/type: geometric::FMT
 * /test/move_group/planner_configs/KPIECE/border_fraction: 0.9
[... 328 lines cut ...]
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1721838257.239529017]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1721838257.239583422]: MoveGroup context initialization complete

You can start planning now!

[ INFO] [1721838258.424786705]: Ready to take commands for planning group xarm6.
^C[test/rviz_Ubuntu20_9361_7429938474669580425-7] killing on exit
[test/move_group-6] killing on exit
[test/robot_state_publisher-5] killing on exit
[test/joint_state_publisher-4] killing on exit
[test/xarm/traj_controller_spawner-3] killing on exit
[WARN] [1721838259.204696]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

nitish3693, 2024-07-24:
Please note : Starting the `xarm6_server.launch` under a ROS_NAMESPACE works,

Logs: 

```
Ubuntu20:~/ros_wk$ ROS_NAMESPACE=test roslaunch xarm_bringup xarm6_server.launch robot_ip:=192.168.1.212
... logging to /home/nitish/.ros/log/29b359fc-49d8-11ef-be2f-c9a005e8bc99/roslaunch-Ubuntu20-8630.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Ubuntu20:34331/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /test/xarm/DOF: 6
 * /test/xarm/add_gripper: False
 * /test/xarm/baud_checkset: True
 * /test/xarm/default_gripper_baud: 2000000
 * /test/xarm/enforce_limits: True
 * /test/xarm/joint_names: ['joint1', 'joint...
 * /test/xarm/prefix: 
 * /test/xarm/uf_model: XARM
 * /test/xarm/velocity_control: False
 * /test/xarm/xarm_report_type: normal
 * /test/xarm/xarm_robot_ip: 192.168.1.212

NODES
  /test/xarm/
    uf_driver (xarm_api/xarm_driver_node)

auto-starting new master
process[master]: started with pid [8638]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 29b359fc-49d8-11ef-be2f-c9a005e8bc99
process[rosout-1]: started with pid [8648]
started core service [/rosout]
process[test/xarm/uf_driver-2]: started with pid [8655]
[ INFO] [1721837823.783537738]: namespace: /test/xarm
[ INFO] [1721837823.785952677]: xarm_driver_node start
SDK_VERSION: 1.13.21
Tcp control connection successful
change protocol identifier to 2
ROBOT_IP: 192.168.1.212, VERSION: v2.3.0, PROTOCOL: V1, DETAIL: 6,6,XI1300,AC1300,v2.3.0, TYPE1300: [1, 1]
Tcp Report Rich connection successful
change protocol identifier to 3
Tcp Report Norm connection successful
report_data_size: 495, size_is_not_confirm: 0
[ INFO] [1721837824.106527657]: [TCP STATUS] CONTROL: 1, REPORT: 1
report_data_size: 145, size_is_not_confirm: 0
^C[test/xarm/uf_driver-2] killing on exit
[xarm_driver] Ctrl-C caught, exit process...
[rosout-1] killing on exit
^C[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete
done
```

nitish3693, 2024-07-25:
@penglongxiang, 

Looks like the error is comming when launching with the argument `use_moveit` set as `true`

penglongxiang, 2024-07-29:
@nitish3693, it seems the namespace and remapping had some issues working together, we are looking into this and will get back to you later.

penglongxiang, 2024-07-31:
Hi @nitish3693, we are still in the process of optimizing the namespace compatibility, you may try with the following launch file temporarily, there are still some remappings needed: 
```xml
<?xml version="1.0"?>
<launch>
  <arg name="ns" default="" doc="namespace as a group"/>
  <arg name="robot_ip" />

  <group ns="$(arg ns)">
    <!-- <remap from="/joint_states" to="/$(arg ns)/joint_states" /> -->
    <remap from="/robot_description" to="/$(arg ns)/robot_description" />
    <remap from="/move_group/monitored_planning_scene" to="/$(arg ns)/move_group/monitored_planning_scene" />
    <remap from="/$(arg ns)/source_list" to="/source_list" />
    <include file="$(find xarm7_moveit_config)/launch/realMove_exec.launch">
      <arg name="robot_ip" value="$(arg robot_ip)"/>
    </include>
  </group>

</launch>
```
If you launch this file and specify `ns:=test` in the command line, you may achieve the similar namespace isolation. We will update the repository after we have finished the optimization.

nitish3693, 2024-08-02:
Hi @penglongxiang, we tested the launch file, 

The initial moveit plan work but the current state of the robot is not updating, causing consequent palns to have in correct start position.

[image]

Current State in Moveit : Light white arm

Current State of Real Hardware : Solid white arm.

penglongxiang, 2024-08-05:
Hi @nitish3693, please check your output of 
`rostopic list | grep monitored`
command and the `Planning Scene Topic` in your [moveit.rviz](https://github.com/xArm-Developer/xarm_ros/blob/7e5752bf36322a8d949610affcd484a115e73d6c/xarm6_moveit_config/launch/moveit.rviz#L140) configuration file in the `xarm6_moveit_config` package (if this is the corresponding launch package).

nitish3693, 2024-08-05:
Hi @penglongxiang 

output of command, 
```
@Ubuntu20:~/ros_wk$ rostopic list | grep monitored
/test/move_group/monitored_planning_scene
```

In the moveit.rviz : 

```
Planning Scene Topic: move_group/monitored_planning_scene
```

nitish3693, 2024-08-05:
@penglongxiang  note in the `xarm7_moveit_config`, `moveit.rviz`,   the planning scene topic is using absolute path,

```
Planning Scene Topic: /move_group/monitored_planning_scene
```

nitish3693, 2024-08-05:
I will test by changing that to relative path and will check.

vimior, 2024-08-06:
@nitish3693 
I pushed an update to the master branch. Can you try to see if it solves your current problem?

nitish3693, 2024-08-16:
@penglongxiang  @vimior  thanks for the update. 

We were able to run xarm6 and xarm7 on same machine, but the tf was completely merged,

[image]