xArm-Developer/xarm_ros #237: hand eye calibration with eye to hand

Murali25111993, 2024-07-29:
Hello,

I am using xarm6 i tried xarm6_vision [Link](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#7-xarm-vision) with eye_on_hand, it was working fine. 

But, I want to test with eye_to_hand. I have attached the image of the setup, I measured the xyz of camera with respect to /link_base it was  x = 0.53 , y =-0.213 z = 0.897, I am finding difficulty in setting the quaternion, 

[image]

[image]

if I made changes in calibrate.launch to include quaternion as below, **marker_frame** is giving wrong value

`<node unless="$(arg eye_on_hand)" name="dummy_handeye" pkg="tf" type="static_transform_publisher"
            args="0.53 -0.213 0.897 1 0 0 0 $(arg robot_base_frame) $(arg tracking_base_frame) 10" />`

[image]

Kindly assist in this. Will share additional files if required

Thanks in advance.

penglongxiang, 2024-07-30:
Hi @Murali25111993, your measured relationship should be the result of this calibration (from robot base to camera frame), the exact transformation is unknown during the process. So there is no need for you to publish this static `tf` relationship. 

As long as you specify "`eye_on_hand`" as `false` in the [d435i_xarm_auto_calib.launch](https://github.com/xArm-Developer/xarm_ros/blob/7e5752bf36322a8d949610affcd484a115e73d6c/xarm_vision/d435i_xarm_setup/launch/d435i_xarm_auto_calib.launch#L58) file, the calibrate.launch from easy_handeye will automatically publish a **dummy** relationship for you to make the `tf` tree complete during calibration sampling process, refer the script [here](https://github.com/IFL-CAMP/easy_handeye/blob/ffcc43d1e63cd16b3052f499c681cd8fbc7b71f5/easy_handeye/launch/calibrate.launch#L37). 

You may just remove your own static publisher and finish a whole round of calibration, **ignoring the fake relationship** displayed in rviz before the final calculation, and compare the final result to your measured values.

Murali25111993, 2024-07-30:
Thanks @penglongxiang . I will try and update you.

Murali25111993, 2024-07-30:
Hi,

I have tested and able to obtain fair results.

 

> parameters:
  eye_on_hand: false
  freehand_robot_movement: true
  move_group: xarm6
  move_group_namespace: /
  namespace: /xarm_depthvista_handeyecalibration_eye_on_base/
  robot_base_frame: link_base
  robot_effector_frame: link_eef
  tracking_base_frame: camera_color_optical_frame
  tracking_marker_frame: camera_marker
transformation:
  qw: -0.0690549
  qx: 0.837
  qy: -0.522232
  qz: -0.148109
  x: 0.535297
  y: -0.213
  z: 0.897

Next step will be object detection and grasping. Closing this issue,

Thanks

closed 2024-07-30 by Murali25111993