xArm-Developer/xarm_ros #238: Vision guided grasping demo - after calibration - no tf link between /object and camera

Murali25111993, 2024-07-30:
Hello @penglongxiang ,

After completing the calibration, now I moved to [vision guided grasping demo](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#73-vision-guided-grasping-demo). 

I made necessary changes as mentioned in the readme file by changing the calibrated yaml file and updated the publish_tf.launch with new file name.  

Made changes to **start_find_obj_2d.launch** by replacing my camera topic name. 

find_2d_object GUI islaunching, I could able to add new object and it is successfully detecting.

[image]

But my tf is having a problem as it says there is no link between /object_3 and frame[world].

**Note:** I didn't publish any tf , which says rgb_frame, it should be launched by **find_object_2d**

[image]

I did 
`rosrun rqt_tf_tree rqt_tf_tree`

[image]

as you can see there is no link between camera_optical_frame or link_base with rgb_frame or /object_3.

whether I need to run any static transform ?

Thanks

Murali25111993, 2024-07-30:
Update!

When I publish static tf between /camera_optical_frame and /rgb_frame. I could able to generate complete tree.

`rosrun tf static_transform_publisher 0 0 0 0 0 0 camera_color_optical_frame rgb_frame 10
`

But , I am getting lot of error like below:

> rosrun d435i_xarm_setup findobj2d_grasp_moveit
[ INFO] [1722343557.777460166]: Plan SUCCESS! Executing... 
[ERROR] [1722343557.888770354]: Failed to call service exec_plan
[ WARN] [1722343558.379358396]: recognition listener: x: 0.352527, y: -0.062724, z: 0.184165, q: [-0.084694, 0.659489, 0.260264, 0.700117]
[ INFO] [1722343558.476887399]: Plan SUCCESS! Executing... 
[ERROR] [1722343560.234317180]: Failed to call service exec_plan
[ INFO] [1722343561.276938493]: Plan SUCCESS! Executing... 
[ERROR] [1722343561.393263857]: Failed to call service exec_plan
[ INFO] [1722343562.477099176]: Plan SUCCESS! Executing... 
[ERROR] [1722343564.219283967]: Failed to call service exec_plan

Kindly tell me the what should be done. !. I even tried adding wait_for_service still same issue persist. 

Thanks

penglongxiang, 2024-08-01:
I tested the application with eye-on-hand and it worked fine:
```
[ INFO] [1722484578.289452287]: Plan SUCCESS! Executing... 
[ WARN] [1722484588.221498042]: recognition listener: x: 0.317007, y: -0.317107, z: 0.091406, q: [-0.130370, 0.665205, 0.105819, 0.727535]
[ INFO] [1722484588.306659405]: Plan SUCCESS! Executing... 
[ INFO] [1722484591.008308104]: Plan SUCCESS! Executing... 
[ INFO] [1722484593.307133461]: Plan SUCCESS! Executing... 
```
If `xarm_exec_plan` service can be listed out, please check if xArm is not in a proper mode (1), it can be checked by: 
`$  rostopic echo /xarm/xarm_states`
If the `mode` field is not 1, or `err` filed is not 0. Then the robot can not respond to Moveit commands. Please clear the error if any and restart the launch process.

penglongxiang, 2024-08-01:
Also, I did not find the link `/rgb_frame` during my latest test, so I do not have to publish any additional `tf` relationship, it is kind of strange.
[image]
[image]

As you can see, `object_2` is and `camera_color_optical_frame` is directly related by the find_object result.

Murali25111993, 2024-08-01:
Ok, Thanks for detailed info. I will check and update you.

closed 2024-09-12 by Murali25111993