xArm-Developer/xarm_ros #239: Using XArm6 with xArm Api and Online Target Update

Akumar201, 2024-09-05:
Hello, I would like to know can there be any issue , controlling the arm in  [Online Target Update mode](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#6-online-target-update) and using xarm API ki to an emergency kill the arm together, last time when I tried doing something like this  I had conflicts between moveit and xarm api . 

Any help will be much appreciated.

penglongxiang, 2024-09-10:
Hi @Akumar201 we do not quite understand your issue. Could you explain why you would like to use online target update and moveit together? And what is exactly the conflict? Any reproducible scripts or detailed commands to show the issue will be appreciated.

Akumar201, 2024-09-10:
Hello @penglongxiang , I am sorry about the explanation earlier , I wanted to use the [Online Target Update mode](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#6-online-target-update) for tracking an object and want to have a functionality button on my user interface to kill the arm in case of emergency, earlier I was using moveit to track the object and xarm API to kill the arm or do emergency stop or manual mode and it had some ownership issues, I just wanted to know if I can do it in the same code without interfering the ownership of the arm. Because , as far I know [Online Target Update mode](https://github.com/xArm-Developer/xarm_ros/tree/master/examples#6-online-target-update) uses ros services to send command to the robotic arm , but for emergency kill or manual mode using the below code uses xarm API, so there might be problem with the ownership . 
```

void XArmWrapper::EmergencyStop()
{
  if (arm != nullptr)
  {
    int ret = arm->motion_enable(false);
    }
    }
```

Is there a way using ROS service I can do motion_enable to false ? Please let me know if the question is clear to you now . Any help would be much appreciated

penglongxiang, 2024-09-11:
Sure there is a ros-service way to disable the motor. Refer to [this part](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#starting-xarm-by-ros-service) of ReadMe. Disabling the motors can be done by:
```shell
$ rosservice call /xarm/motion_ctrl 8 0
```
First argument is target joint ID: (1-7) for single joint and 8 for all joints. Second argument: 1 for enabling and 0 for disabling.

Akumar201, 2024-09-11:
@penglongxiang Thank You so much for your help !

closed 2024-09-11 by Akumar201