xArm-Developer/xarm_ros #240: How can I extract the urdf of xarm6 with gripper for other emulator

para133, 2024-09-25:
All tutorial for simulation are based on Gazebo and rviz. If I want to use other emulator, how can I extract the urdf and srdf of xarm6 with Two-finger gripper or which xacro should I use ? Looking forward to your reply.

penglongxiang, 2024-09-27:
You may use this command:
```shell
cd xarm_ros
rosrun xacro xacro xarm_description/urdf/xarm_device.urdf.xacro robot_type:=xarm dof:=6 > xarm6_with_gripper.urdf
```

para133, 2024-09-28:
@penglongxiang Thanks for your reply ,so if I want to extract the corresponding srdf ,should I use following command?
```
cd xarm_ros  
rosrun xacro xacro xarm_description/srdf/xarm.srdf.xacro robot_type:=xarm dof:=6 add_gripper:=true > xarm6_with_gripper.srdfk
```

para133, 2024-09-28:
@penglongxiang Thanks for your reply ,so if I want to extract the corresponding srdf ,should I use following command?
```
cd xarm_ros  
rosrun xacro xacro xarm_description/srdf/xarm.srdf.xacro robot_type:=xarm dof:=6 add_gripper:=true > xarm6_with_gripper.srdf
```