xArm-Developer/xarm_ros #247: Combine RVIZ controller and rosservice controller

khuechuong, 2024-12-22:
```roslaunch xarm_planner xarm_planner_realHW.launch``` controls the xarm via rosservice.

```roslaunch xarm5_moveit_config realMove_exec.launc``` controls the xarm via rviz. 

Is there a way I can have launch file that is a combination of ```roslaunch xarm5_moveit_config realMove_exec.launch``` and ```roslaunch xarm_planner xarm_planner_realHW.launch```? In case I can't use rosservice, I can use rviz to move joints.

penglongxiang, 2024-12-24:
Actually both the specified launch files use `Moveit` planner and `ros-controller` to control xArm. The launch files from `xarm_bringup` use services provided by `xarm_api` package, if this is what you are referring to.

While it is **not recommended** to use the two methods combined at the same time, since they operate in different mode of the robot, **mode 1** for moveit and mostly **mode 0** for the service, and the states and commands may have **conflicts** if not programmed properly. 

The service servers are actually also launched in your mentioned launch files (for moveit), the switching between the two control method requires mode change and then setting state to 0. **It is important to read through our document and do small step command test between the two methods before you actually do so in your application, making sure there is no sudden jump back upon mode switching to ensure safety.**  Also, please update your xarm_ros code and controller firmware to the latest version to make sure the known bugs has been fixed.

khuechuong, 2024-12-29:
```roslaunch xarm5_moveit_config realMove_exec.launch``` I didn't see any rosservice that could receive a pose and plan end-defector to the position.

penglongxiang, 2025-01-13:
@khuechuong Please update to the latest code and try:
```
$ roslaunch xarm_planner xarm_planner_realHW.launch robot_ip:=<your_robot_ip> robot_dof:=5 no_gui_plan:=false
```
By setting `no_gui_plan:=false` will enable the interactive marker and motion planning panel in Rviz, you can use both the service and Rviz GUI to do motion planning. But remember to click the **Start State** to be **current** in the MotionPlanning panel->Planning->Query again after the switch from service to GUI, or the start state may not be synchronized to rviz and error may occur upon execution.

khuechuong, 2025-01-15:
@penglongxiang 
So I ran ```roslaunch xarm_planner xarm_planner_realHW.launch```
Then I used ```xarm_pose_plan``` and ```xarm_exec_plan``` to move, then I ran
```
$ rosservice call /xarm/set_mode 0
$ rosservice call /xarm/set_state 0
```

To be able to run:
```
rosservice call /xarm/move_joint [0,0,0,0,0] 0.35 7 0 0
```
Afterwards, I can't use ```xarm_pose_plan``` and ```xarm_exec_plan``` again. How do I set it so I can use it again?

penglongxiang, 2025-01-16:
@khuechuong Since xarm_planner is using moveit in the background, and it should be operating under Mode 1. If you wish to switch the control from mode 0 to moveit planner, a faster way is to call:
```
$ rosservice call /xarm/moveit_clear_err
```
As was mentioned [here](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#clearing-errors). Please check if the planned trajectory and starting state are valid before execution after the switch, and always keep emergency stop within your reach.

khuechuong, 2025-01-16:
@penglongxiang
So I ran roslaunch xarm_planner xarm_planner_realHW.launch
Then I used xarm_pose_plan and xarm_exec_plan to move. Is there an option to change velocity scaling? or do I just change in joint_velocity.yaml?

Also, it seems a too sensitive to resistant, how do I make it less sensitive?

closed 2025-01-17 by khuechuong