xArm-Developer/xarm_ros #248: Disable Collision that stops the xArm

khuechuong, 2025-01-16:
Is there a way to disable the collision estop for a task then re enable it? I'm doing something similar to a sanding problem where the xArm has to make contact. The margin of error is thin so I want to disable it while it press and pull back. Then re-enable it.

penglongxiang, 2025-01-17:
Please use the service "`xarm/set_collision_sensitivity`" to modify the collision sensitivity level. The checking range is from 1 to 5, from least sensitive to most sensitive, 3 is default level. To completely **disable** the collision check, set this sensitivity level to **0**, please make a risk assessment before disabling the collision check.  

Please note you have to set state to 0 again before any motion command after the sensitivity change.

khuechuong, 2025-01-17:
@penglongxiang  is there a way to set it before hand? I want to changing it 1 would work.

penglongxiang, 2025-01-20:
You can do it with **UFactory Studio** (also available by accessing `http://your_robot_ip:18333` in web browser), go to Settings(gear logo)->Motion->Parameters->Collision Detection->Sensitivity, choose your desired level then click "**Save**" button, then this configuration will be saved even after system reboot.

closed 2025-02-17 by khuechuong