xArm-Developer/xarm_ros #250: switching between moveit planner and moveit servos

khuechuong, 2025-03-11:
So I want to use both functionality of 

the jogging control from ```xarm_moveit_servo_realmove.launch``` by publishing to rostopic ```/servo_server/cartesian_command_in_topic```

and 

the planner and execute of pose from ```xarm_planner_realHW.launch``` by calling rosservice  /xarm_pose_plan and /xarm_exec_plan

how do I create a launch file that does that? and how would changing modes and state works switching between?

khuechuong, 2025-03-13:
@penglongxiang

penglongxiang, 2025-03-14:
please check my answer under [this issue](https://github.com/xArm-Developer/xarm_ros2/issues/117#issuecomment-2700371920).

khuechuong, 2025-05-02:
@penglongxiang 
So basically, since I use Noetic, I should launch ```xarm_planner_realHW.launch```, then rosrun the ```xarm_keyboard_to_servo_node```? Then switch between mode 1 and 0?

penglongxiang, 2025-05-08:
Sorry I made a mistake, I thought you were to combine `xarm_moveit_servo` with `xarm_api` services. Because those two operates in different mode, it is possible to switch between the two ways.

If you wish to combine `xarm_moveit_servo` and `xarm_planner`, I do not think there is a fast way to implement it. Because they basically operate in the same robot mode (1) with Moveit + ros-controller running behind, they will conflict with each other (the real state and command) I think.

khuechuong, 2025-05-08:
so there another way? I want the ```servo_node``` b/c I want the arm to face an angle, then go straight to that angle rather than planning a bunch of waypoints. But I need the ```xarm_palnner_realHW.launch``` b/c I need to move the arm around.

khuechuong, 2025-05-08:
@penglongxiang

penglongxiang, 2025-05-09:
Is it possible to use **ROS2** and combine `moveit_servo` with `xarm_api` (Cartesian and joint space motion available) service as described in the [previous link](https://github.com/xArm-Developer/xarm_ros/issues/250#issuecomment-2724214215)? In ROS1 I can not add the same controller deactivation logic when mode switches (to prevent sudden jump back to previous moveit servo command pose), because `controller_manager/switch_controller` service is blocking forever and I have not seen the cause yet. So in ROS noetic, it is still not safe to switch mode when `moveit_servo` is operating. 

ROS Noetic will be EOL next month so It is better to use ROS2 which will be actively developed and more flexible.If ROS2 is not an option, I have no better workaround so far. It is better to stick to one single control source. You may also do additional researches and add your own logic, but always be **cautious** for the command conflict and **do enough risk assessment**, use small/slow motions when testing and always have emergency stop at reach.

penglongxiang, 2025-05-13:
@khuechuong, today I found after **upgrading** `ros-noetic`, `ros-control`, `ros-controllers` and `moveit` **packages to the latest version** (mostly updated in **May 2025**). The switching control between `moveit-servo` and `xarm_api` services (in mode 0) will not cause the robot to jump back. I guess in previous version I used (updated May 2024), `moveit_servo` were not actively updating the current state from hardware interface, now it has been optimized. You may have a try.

After upgrading you can check the version by something like `dpkg -l | grep ros-noetic-moveit-servo`, the version info will come up with a date like `20250503`.

Then just launch moveit_servo demo, it will also bring up the api services. For example, If using keyboard control, the joystick type is 99,  it will be like:
```bash
$ roslaunch xarm_moveit_servo xarm_moveit_servo_realmove.launch robot_ip:=192.168.1.127 dof:=7 joystick_type:=99 
```
Then you may set the mode to 0 to use our internal planning services:
```bash
$ rosservice call /xarm/set_mode 0
$ rosservice call /xarm/set_state 0
```
Then, for Cartesian straight line motion (Note: Cartesian Unit is in **mm** and radians for API service): 
$ rosservice call /xarm/move_line [250,100,300,3.14,0,0] 200 2000 0 0 
Please check the [ReadMe](https://github.com/xArm-Developer/xarm_ros?tab=readme-ov-file#57-xarm_apixarm_msgs) for more detailed information. 

If you plan to use moveit_servo again, switch back to mode 1 and state 0:
```bash
$ rosservice call /xarm/set_mode 1
$ rosservice call /xarm/set_state 0
```
or Simply:
```bash
rosservice call /xarm/moveit_clear_err 
```
Please be careful when you try this control switch in the first time, if making **a small relative motion** in Mode 0 and it does not jump back after switching back to mode 1 (state 0), and you can control the robot live with keyboard again with no problem, then it will be OK for you to combine the two control method with only one launch file above.

khuechuong, 2025-05-13:
@penglongxiang would it still be able to moveit to specific location if I give it a pose? because I need a specific position to move to before doing the cartesian straight line motion.

lu4k87, 2025-05-28:
U can set your Pose here: 

$ rosservice call /xarm/move_line **[250,100,300,3.14,0,0]** 200 2000 0 0 
right? @penglongxiang 

I will be try next days and will report here again..
Its exactly what if searched for, thanks!